FrameworkComputer/ExpansionBay #2: Estimated thickness, External ports?

maxtim10, 2023-04-02:
Need the dimensions and details of not just PCBA but the housing.  Planning to start development work on an industry-specific expansion module, but I need to at least know the rough dimensions to determine feasibility.  

In the Framework announcement, it was stated thickness can be whatever, but I'm sure the PCBA will need a connection port to "slot in" the mainboard.

Another point that I need clarification on is will the housing be all the way at the back of the laptop for potentially additional port additional ports, etc.

NickelLiang, 2023-04-03:
Also want to have more details about the dimensions of the housing, also planning on an industry-specific expansion module. 

We already know that our expansion module definitely will not fit in the standard Expansion Bay Shell, so a reference design for Graphics Module / no Z-height limitation module will be of great help for us to evaluate the feasibility of implementing our solution as a module for Framework 16. 

Thanks!

Cutieguwu, 2024-01-23:
Going to quickly bump this.

Details regarding the housing design and base construction are very important for custom module design. Expanding the housing from a base model is easy if you don't have to reconstruct all of the housing in contact with the laptop beforehand (Connectors, rails, etc). Also helpful if you plan to produce a module for a shell before you have the laptop on hand. Then you can at least begin the initial design and build stage before proper testing.